Xmegaduino/Xmegaduino #40: Internal Oscillator Loses Accuracy at Lower Temps

lthiery, 2015-04-28:
I've recently fixed a bug that had been chasing me on the Xmegas for quite a while. The symptom was that my Serial communications (running off of the internal oscillators) would lose themselves as temperatures went below 20C. Sometimes it would be at 14C, sometimes at 4C, sometimes at -2C. I was deploying in cold places and invariably, the cold would ruin my application.

Well, I found a code snippet that helps. Perhaps it should be the defaults when a board is running off the internal? I don't really know. Anyway, here it is in case this helps someone.

```
 OSC.CTRL |= OSC_RC32MEN_bm | OSC_RC32KEN_bm;  /* Enable the internal 32MHz & 32KHz oscillators */
  while(!(OSC.STATUS & OSC_RC32KRDY_bm));       /* Wait for 32Khz oscillator to stabilize */
  while(!(OSC.STATUS & OSC_RC32MRDY_bm));       /* Wait for 32MHz oscillator to stabilize */
  DFLLRC32M.CTRL = DFLL_ENABLE_bm ;             /* Enable DFLL - defaults to calibrate against internal 32Khz clock */
  CCP = CCP_IOREG_gc;                           /* Disable register security for clock update */
  CLK.CTRL = CLK_SCLKSEL_RC32M_gc;              /* Switch to 32MHz clock */
  OSC.CTRL &= ~OSC_RC2MEN_bm;                   /* Disable 2Mhz oscillator */```

karlbackstrom, 2015-05-18:
Sorry for the late reply, will check this next week when I am back from Shenzhen.
Will also plan a new release now that Arduino has released the Board Manager.

karlbackstrom, 2015-05-30:
CAn you make a pull-request for this issue? Notice that I changed the organization for this repo so you probably want to update your remotes. "git remote set-url..."

lthiery, 2015-06-08:
Well... I'm not sure where exactly we'd want to put this. Do you have any ideas?

Otherwise, let me put some thought into where it should go and when I have an idea, I'll issue a pull request in a month or so.

karlbackstrom, 2015-06-09:
Not right now, so takt the time you need. :)

KHeintz, 2015-06-18:
Wiring.c must be changed following. Include tuntime calibration and fix the RTC

/*
  wiring.c - Partial implementation of the Wiring API for the ATmega8.
  Part of Arduino - http://www.arduino.cc/

  Copyright (c) 2005-2006 David A. Mellis

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General
  Public License along with this library; if not, write to the
  Free Software Foundation, Inc., 59 Temple Place, Suite 330,
  Boston, MA  02111-1307  USA

  $Id: wiring.c 808 2009-12-18 17:44:08Z dmellis $
*/

#include <stddef.h>
#include <avr/pgmspace.h>
#include "wiring_private.h"

/* Set internal 32.768kHZ oscillator as clock source for RTC. */
#define CLK_RTCSRC_RCOSC32_gc (0x06<<1)
//CLK_RTCSRC_TOSC32_gc = (0x05<<1)

// Some versions of avr-gcc on linux defines XXX_bp instead of _gp.
#if !defined ADC_SWEEP_gp
    #define ADC_SWEEP_gp  ADC_SWEEP0_bp
#endif
#if !defined ADC_EVSEL_gp
    #define ADC_EVSEL_gp  ADC_EVSEL0_bp
#endif
#if !defined ADC_EVACT_gp
    #define ADC_EVACT_gp  ADC_EVACT0_bp
#endif
#if !defined ADC_CH_GAINFAC_gp
    #define ADC_CH_GAINFAC_gp  ADC_CH_GAIN_1X_gc
#endif
#if !defined ADC_DMASEL_gp
    #define ADC_DMASEL_gp  ADC_DMASEL0_bp
#endif

volatile unsigned long millis_count = 0;
volatile unsigned long seconds_count = 0;
#if defined(USE_RTC)
volatile unsigned long rtc_millis = 0;

/*! \brief RTC overflow interrupt service routine.
 *
 *  This ISR keeps track of the milliseconds 
 */
ISR(RTC_OVF_vect)
{
	rtc_millis = rtc_millis+4;
}

unsigned long millis()
{
 	unsigned long m;
 	
 	uint8_t oldSREG = SREG;
[... 386 lines cut ...]
{
    NVM_CMD = NVM_CMD_READ_CALIB_ROW_gc;
    uint8_t result = pgm_read_byte(index);
    NVM_CMD = NVM_CMD_NO_OPERATION_gc;

    return result;
}
#endif // ADCACAL0 || ADCBCAL0

// TODO: Do this only for xplain board.

static const byte resetPin = 31;

static void resetRupt() {
    if ( 1 == digitalRead(resetPin) ) {
        CCP = CCP_IOREG_gc;
        WDT.CTRL = WDT_PER_8CLK_gc
                 | WDT_ENABLE_bm
                 | WDT_CEN_bm;
    }
}

static void initResetButton() {
    attachInterrupt( resetPin, resetRupt, FALLING );
}