Xmegaduino/Xmegaduino #44: ADC Voltage Reference Selection variable naming

zdjurisic, 2017-04-30:
There seem to be an issue with naming of the ADC Vref variables in wiring.c, line 392. Out of 45 Xmega MCUs in MCU .h files supplied with Arduino 1.8.2 (avr-gcc 4.9.2, avrdude 6.3.0) 15 use ADC_REFSEL_VCC_gc, and 30 use ADC_REFSEL_INTVCC_gc for "Internal VCC/1.6". Introducing any of the latter MCUs would require not just pins definition, but wiring.c modification, too.

zdjurisic, 2017-05-01:
Since nobody is responding, here is what would work: in wiring.c, the code in lines 392-395 should be replaced by:

`#if defined(ADC_REFSEL_VCC_gc)
        adc->REFCTRL = ADC_REFSEL_VCC_gc   // VCC/1.6 analog ref
                     | 0 << ADC_BANDGAP_bp // bandgap not enabled
                     | 0 << ADC_TEMPREF_bp // temerature reference not enabled
                     ;
#endif
#if defined(ADC_REFSEL_INTVCC_gc)
        adc->REFCTRL = ADC_REFSEL_INTVCC_gc   // VCC/1.6 analog ref
                     | 0 << ADC_BANDGAP_bp // bandgap not enabled
                     | 0 << ADC_TEMPREF_bp // temerature reference not enabled
                     ;
`

That seems to cover analog reference for all 45 XMegas included in avr-gcc supplied with current Arduino IDE (1.8.2). If anybody is willing to make a pull request out of this, I'd appreciate it, since I am not sure how to do that myself, and would avoid learning how since too busy on the application side.

karlbackstrom, 2017-05-01:
Hi,

Thank you for the patch, will look at it during the week and try to put it in.

Regards,
Karl